etchdroid/etchdroid #278: Upload via usb error

Hamsavn916, 2025-04-09:
There was an error

The USB device content does not match the image

If you can, please report this issue on GitHub along with the steps that led to this error.

View GitHub issues

Try again

Start over

depau, 2025-04-09:
Hi! Can you further explain what you did to get the error?

Anonymous2716, 2025-04-11:
I believe I have the same issue. I'm on HyperOS 2 (Android 15).  shows the exact message right after finishing writing the iso.

edi20psulex, 2025-04-23:
I also have the same problem and need a solution fast!

depau, 2025-05-02:
I'm aware of this issue. For now the workaround is to just try again. Does this happen every time?

> I also have the same problem and need a solution fast!

Hi! If you need a solution fast you can reach out via email and I can send you an invoice for fixing the issue on contract.

Anonymous2716, 2025-05-15:
> Does this happen every time?

 yes.

BlackwiRe-live, 2025-11-07:
Same here on a s25 image dont match which dont make no damn sense then what was downloaded Chinese Spyware lol

pzb568, 2026-10-22:
The new version has issues. You have to go back to version 1.51.