etchdroid/etchdroid #281: Feature Request: Create Multiboot USB Media

just-Nob, 2025-04-24:
## Introductory Foreword:

First of all, thanks for that cool tool!
With EtchDroid v.2.0, it clearly seems to work much more smoothly for me, and I appreciate the possibility of writing ISO images even if there's no working PC available!

## Description:

When writing OS installation ISO images to bootable USB sticks / SDCards in the past, I liked to use YUMI for creating multiboot USB media.

This offered two major advantages:
- Multiple ISO images to boot on one single media (for example, particularly useful for testing different Linux distributions or for different rescue images),
- possibility to additionally use the free/unused media's memory as "usual" USB data storage.

For these reasons, it would be really great if EtchDroid would also support this feature/functionality in future versions!

depau, 2025-05-02:
Hi! Thanks for your suggestion. This has been requested quite a few times but my plan for EtchDroid is to keep it as simple as possible and to make it do only one thing and do it well.

Creating multiboot media is an involved process and it requires a high degree of complexity that I'm not willing to introduce. I would also be very reluctant to accept contributions from other people towards this front.

closed 2025-05-02 by depau

just-Nob, 2025-05-02:
Hi, thanks for the answer.
OK, got it. Anyway, it was worth the request. :)