etchdroid/etchdroid #295: Writing the ISO Gets Stuck

Dharma-kdk, 2025-06-07:
Every time i try to put the iso in the usb with etch droid a error message says that the writing process is stuck, 

I tried 2 linux distro and is the same so i assume is not about the distro and may be something about app compability cause some times the app has troubles even before the "stuck"  error appears things like crashing or redirecting me to the menu of the app all of this happens with the same steps pressing the button to write the image onto the usb select the iso file and then or loads and "stuck" or crashes or just back to the main menu

(yeah english is not mi main language sorry about errors or lack of technical language or if is hard to understand me)

depau, 2025-06-17:
@technicallearnings that's really really really interesting. Can you attach here the mtp host app manifest and detail the changes you had to make? 

By the way the app is free open source software, no need to decompile it :)

depau, 2025-07-06:
Hey @technicallearnings, just a ping. Any chance you can send the mtp host app you mentioned?

BiNGT52, 2025-07-06:
> @BiNGT52 / @Dharma-kdk / @technicallearnings what devices are you using?

I tried it on Poco m5s, and Samsung galaxy A02

technicallearnings, 2025-07-06:
@depau @BiNGT52 Hey this was my first ever comment on GitHub I just saw your bump in my email I'm not to used to this so I apologize for my lateness. Im on a xiaomi redmi 13 pro 4g (sapphire) at the time I believe I was using crdroid as a custom ROM but I'm on evolution x now. I don't think the roms make a difference and I've since lost that app since my ROM switch but I'll re do the process real quick and hand over that xml, also I'll try to attach the modded APK if any one wants. I was having the same (almost same) issue with drivedroid as well but some added permissions from a couple other other apks seemed to fix it so i don't even use my USB drive anymore I just use my phone to boot whatever programs with drivedroid. Anyways, Ill upload that stuff here in the next 10 minutes or so.

technicallearnings, 2025-07-06:
@depau [image]

depau, 2025-07-07:
Thank you! No worries, I just pinged you since I ended up in this post anyway. I don't know when I'll have time to look at it, I usually just wait until a few things to do bunch up and then work on them for some time.

Thanks for all the info.

BiNGT52, 2025-07-08:
@technicallearnings  

Hi :) Thanks for xml and apk, but idk what i have to do with this...

depau, 2025-07-08:
@BiNGT52 don't worry, it's for me

Ben-bit-code208, 2025-12-17:
look here that maybe works #328